battosai30/MCP3424 #1: math.h

jano2358, 2015-04-04:
When MCP 3424 version 1.2 example sketch Multichannel is compiled in IDE 1.6.0 on linux it demands for Math.h (wich is included in MCP3424.h), but fails to found it, it works if i change to math.h.
Why the rest of the word don' care about capital letters?

battosai30, 2015-04-06:
Maybe because my library has not the success it deserves ?? xD

I have no answer, when I wrote it it was before IDE 1.0 I think. Two possibilities : 
- It was Math.h
- There was a #define Math.h math.h

I'll correct it 

Thx

jano2358, 2015-04-06:
well... i cannot have readings anyway :-(
if i put the wrong adress for my device i get 0 microvolt allways, if i put the 'correct' (?) address, i get -74 microvolt on all channels which is impossible since 3 and 4 are unconnected!
i don't know how i can be more usefull to solve this (i'm more on the side of electronics, not programming), right now i'm using this library: https://github.com/codebendercc/arduino-library-files/tree/master/external-libraries/MCP3424
it works even i cannot put the adress properly (i needed to change inside MCP3424.h) or the gain

battosai30, 2015-04-07:
How okay you wrote your own ;)

It's not working because you do not use the right adress : 

 Wire.requestFrom(MCP3424_ADDRESS, 4); in readData()

Wire.beginTransmission(_address); in MCP3424Write()

PR #2 (2017-11-15): Case corrections
A couple of case corrections to fix #1
closed 2017-11-15 by battosai30