battosai30/MCP3424 #5: Does not compile under actual arduino version

red-gee, 2019-07-24:
Could you tell me what version of arduino ide do I need to compile? 
(  I take your example it puts out an error during compile 
MCP not declared )
### **Or much better:** 

What I have to do to make it run under actual ide?

battosai30, 2019-07-24:
Hi, 

Can you tell me what is the example failing and the line numer error ?

red-gee, 2019-07-24:
Hi, the example _Multichannel.ino
In file included from C:\Users\r\Documents

Arduino IDE Version 1.8.3 here on Windows, but nearly the same error codes on Mac with latest Arduino IDE

\Arduino\libraries\MCP3424\Examples\MCP3424_Multichannel\MCP3424_Multichannel.ino:8:0:

C:\Users\r\Documents\Arduino\libraries\MCP3424/MCP342x.h:42:1: warning: 'typedef' was ignored in this declaration

 };

 ^

C:\Users\r\Documents\Arduino\libraries\MCP3424/MCP342x.h:50:1: warning: 'typedef' was ignored in this declaration

 };

 ^

C:\Users\r\Documents\Arduino\libraries\MCP3424/MCP342x.h:58:1: warning: 'typedef' was ignored in this declaration

 };

 ^

C:\Users\r\Documents\Arduino\libraries\MCP3424/MCP342x.h:66:1: warning: 'typedef' was ignored in this declaration

 };

 ^

MCP3424_Multichannel:11: error: 'MCP3424' does not name a type

 MCP3424 MCP(0); // Declaration of MCP3424 pin addr1 et addr0 are connected to +5V

 ^

C:\Users\r\Documents\Arduino\libraries\MCP3424\Examples\MCP3424_Multichannel\MCP3424_Multichannel.ino: In function 'void setup()':

MCP3424_Multichannel:18: error: 'MCP' was not declared in this scope

   MCP.begin(0);

   ^

C:\Users\r\Documents\Arduino\libraries\MCP3424\Examples\MCP3424_Multichannel\MCP3424_Multichannel.ino: In function 'void loop()':

MCP3424_Multichannel:29: error: 'MCP' was not declared in this scope

     MCP.configuration(i,12,1,1); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8 

     ^

MCP3424_Multichannel:46: error: 'MCP' was not declared in this scope

     MCP.configuration(i,14,1,1); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8 

     ^

MCP3424_Multichannel:63: error: 'MCP' was not declared in this scope

     MCP.configuration(i,16,1,1); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8 

     ^

MCP3424_Multichannel:80: error: 'MCP' was not declared in this scope

     MCP.configuration(i,18,1,1); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8 

     ^

exit status 1
'MCP3424' does not name a type

battosai30, 2019-07-25:
Oh yeah, as I worked on MCP3421, I changeg the lib to MCP342x.h and I changed class name too.

So replace 
`MCP3424 MCP(0); // Declaration of MCP3424 pin addr1 et addr0 are connected to +5V`
By
`MCP342x MCP(0);`

red-gee, 2019-07-25:
thank you for your support, I download the latest version right now, but now there is an other error message.

/var/folders/tc/hw4zyc795rxbrs8rd3s0mzmh0000gn/T/arduino_modified_sketch_983999/MCP3424_Multichannel.ino: In function 'void loop()':
MCP3424_Multichannel:50:9: error: 'class MCP342x' has no member named 'configuration'
     MCP.configuration(i,18,1,1); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8 
         ^
exit status 1
'class MCP342x' has no member named 'configuration'

im on Arduino 1.8.9

battosai30, 2019-07-26:
Change to setConfiguration()
Another thing I have to change xD

red-gee, 2019-07-26:
I’ve tested it with setConfiguration, but the one shot sktetch wont compile

battosai30, 2019-07-26:
I updated OneSHot example on Github, pull the git on my computer and I succesfully compiled on Arduino and STM32 platforms

red-gee, 2019-07-26:
now it works... thank you very much.

closed 2019-07-26 by red-gee