battosai30/MCP3424 #6: example MCP3424_Multichannel doesnt work for only 18bit resolution

avandalen, 2019-10-14:
The example MCP3424_Multichannel doesnt work with only the 18bit resolution part:
`/* MCP 3424 version 1.3 example sketch Multichannel
Written by B@tto 
Contact : batto@hotmail.fr

In this example, the four channel are measured and results displayed by serail connection.
*/

#include <MCP342x.h>
#include <Wire.h>

MCP342x MCP(0); // Declaration of MCP3424 pin addr1 et addr0 are connected to +5V

long Voltage[4]; // Array used to store results

void setup() {

  Serial.begin(9600);  // start serial for output
  MCP.begin(0);

}

void loop(){
/*
  Serial.println(""); 
  Serial.println("12 bits"); 
  
  for(int i=1;i<=4;i++){

    MCP.setConfiguration(i,RESOLUTION_12_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8 

    Voltage[i-1]=MCP.measure(); // Measure is stocked in array Voltage, note that the library will wait for a completed conversion that takes around 200 ms@18bits

    Serial.print("Channel "); // print results
    Serial.print(i);
    Serial.print(" : ");  
    Serial.print(Voltage[i-1]);
    Serial.println(" microVolt");

  }

  Serial.println(""); 
  Serial.println("14 bits"); 
  
    for(int i=1;i<=4;i++){

      MCP.setConfiguration(i,RESOLUTION_14_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8 

    Voltage[i-1]=MCP.measure(); // Measure is stocked in array Voltage, note that the library will wait for a completed conversion that takes around 200 ms@18bits

    Serial.print("Channel "); // print results
    Serial.print(i);
    Serial.print(" : ");  
    Serial.print(Voltage[i-1]);
    Serial.println(" microVolt");

  }

  Serial.println(""); 
  Serial.println("16 bits"); 
  
    for(int i=1;i<=4;i++){

    MCP.setConfiguration(i,RESOLUTION_16_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8 

    Voltage[i-1]=MCP.measure(); // Measure is stocked in array Voltage, note that the library will wait for a completed conversion that takes around 200 ms@18bits

    Serial.print("Channel "); // print results
    Serial.print(i);
    Serial.print(" : ");  
    Serial.print(Voltage[i-1]);
    Serial.println(" microVolt");

  }
  */

  Serial.println(""); 
  Serial.println("18 bits"); 
  
    for(int i=1;i<=4;i++){

    MCP.setConfiguration(i,RESOLUTION_18_BITS,CONTINUOUS_MODE,PGA_X8);// MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8 

    Voltage[i-1]=MCP.measure(); // Measure is stocked in array Voltage, note that the library will wait for a completed conversion that takes around 200 ms@18bits

    Serial.print("Channel "); // print results
    Serial.print(i);
    Serial.print(" : ");  
    Serial.print(Voltage[i-1]);
    Serial.println(" microVolt");

  }
}
`

battosai30, 2019-10-14:
Hi,

"doesnt work" means "not getting the same value as the other resolutions" ?

avandalen, 2019-10-15:
no, nothing happens

From: battosai30 
Sent: Monday, October 14, 2019 8:34 PM
To: battosai30/MCP3424 
Cc: Albert van Dalen ; Author 
Subject: Re: [battosai30/MCP3424] example MCP3424_Multichannel doesnt work for only 18bit resolution (#6)

Hi,

"doesnt work" means "not getting the same value as the other resolutions" ?

battosai30, 2019-10-15:
So not only on 18 bits part ?

avandalen, 2019-10-15:
you can try the code that was included.

From: battosai30 
Sent: Tuesday, October 15, 2019 4:10 PM
To: battosai30/MCP3424 
Cc: Albert van Dalen ; Author 
Subject: Re: [battosai30/MCP3424] example MCP3424_Multichannel doesnt work for only 18bit resolution (#6)

So not only on 18 bits part ?

battosai30, 2020-02-26:
May be kinked to https://github.com/battosai30/MCP3424/pull/7

closed 2020-02-26 by battosai30