battosai30/MCP3424 #8: Compiler errors

hairem, 2021-08-18:
Arduino: 1.8.13 (Windows 10), Board: "Seeeduino XIAO, Arduino, Off"

C:\Users\AMP User\Documents\Arduino\libraries\MCP3424-master\Examples\MCP3424_Multichannel\MCP3424_Multichannel.ino: In function 'void setup()':

MCP3424_Multichannel:18:7: error: 'class MCP342x' has no member named 'begin'

   MCP.begin(0);

       ^~~~~

C:\Users\AMP User\Documents\Arduino\libraries\MCP3424-master\Examples\MCP3424_Multichannel\MCP3424_Multichannel.ino: In function 'void loop()':

MCP3424_Multichannel:29:11: error: 'class MCP342x' has no member named 'setConfiguration'

       MCP.setConfiguration(i,RESOLUTION_12_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

           ^~~~~~~~~~~~~~~~

MCP3424_Multichannel:29:30: error: 'RESOLUTION_12_BITS' was not declared in this scope

       MCP.setConfiguration(i,RESOLUTION_12_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                              ^~~~~~~~~~~~~~~~~~

MCP3424_Multichannel:29:49: error: 'CONTINUOUS_MODE' was not declared in this scope

       MCP.setConfiguration(i,RESOLUTION_12_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                                                 ^~~~~~~~~~~~~~~

MCP3424_Multichannel:29:65: error: 'PGA_X8' was not declared in this scope

       MCP.setConfiguration(i,RESOLUTION_12_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                                                                 ^~~~~~

C:\Users\AMP User\Documents\Arduino\libraries\MCP3424-master\Examples\MCP3424_Multichannel\MCP3424_Multichannel.ino:29:65: note: suggested alternative: 'PRIX8'

       MCP.setConfiguration(i,RESOLUTION_12_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                                                                 ^~~~~~

                                                                 PRIX8

MCP3424_Multichannel:31:22: error: 'class MCP342x' has no member named 'measure'

     Voltage[i-1]=MCP.measure(); // Measure is stocked in array Voltage, note that the library will wait for a completed conversion that takes around 200 ms@18bits

                      ^~~~~~~

MCP3424_Multichannel:46:11: error: 'class MCP342x' has no member named 'setConfiguration'

       MCP.setConfiguration(i,RESOLUTION_14_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

           ^~~~~~~~~~~~~~~~

MCP3424_Multichannel:46:30: error: 'RESOLUTION_14_BITS' was not declared in this scope

       MCP.setConfiguration(i,RESOLUTION_14_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                              ^~~~~~~~~~~~~~~~~~

MCP3424_Multichannel:46:49: error: 'CONTINUOUS_MODE' was not declared in this scope

       MCP.setConfiguration(i,RESOLUTION_14_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                                                 ^~~~~~~~~~~~~~~

MCP3424_Multichannel:46:65: error: 'PGA_X8' was not declared in this scope

       MCP.setConfiguration(i,RESOLUTION_14_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                                                                 ^~~~~~

C:\Users\AMP User\Documents\Arduino\libraries\MCP3424-master\Examples\MCP3424_Multichannel\MCP3424_Multichannel.ino:46:65: note: suggested alternative: 'PRIX8'

       MCP.setConfiguration(i,RESOLUTION_14_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                                                                 ^~~~~~

                                                                 PRIX8

MCP3424_Multichannel:48:22: error: 'class MCP342x' has no member named 'measure'

     Voltage[i-1]=MCP.measure(); // Measure is stocked in array Voltage, note that the library will wait for a completed conversion that takes around 200 ms@18bits

                      ^~~~~~~

MCP3424_Multichannel:63:9: error: 'class MCP342x' has no member named 'setConfiguration'

     MCP.setConfiguration(i,RESOLUTION_16_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

         ^~~~~~~~~~~~~~~~

MCP3424_Multichannel:63:28: error: 'RESOLUTION_16_BITS' was not declared in this scope

     MCP.setConfiguration(i,RESOLUTION_16_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                            ^~~~~~~~~~~~~~~~~~

MCP3424_Multichannel:63:47: error: 'CONTINUOUS_MODE' was not declared in this scope

     MCP.setConfiguration(i,RESOLUTION_16_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                                               ^~~~~~~~~~~~~~~

MCP3424_Multichannel:63:63: error: 'PGA_X8' was not declared in this scope

     MCP.setConfiguration(i,RESOLUTION_16_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                                                               ^~~~~~

C:\Users\AMP User\Documents\Arduino\libraries\MCP3424-master\Examples\MCP3424_Multichannel\MCP3424_Multichannel.ino:63:63: note: suggested alternative: 'PRIX8'

     MCP.setConfiguration(i,RESOLUTION_16_BITS,CONTINUOUS_MODE,PGA_X8); // MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                                                               ^~~~~~

                                                               PRIX8

MCP3424_Multichannel:65:22: error: 'class MCP342x' has no member named 'measure'

     Voltage[i-1]=MCP.measure(); // Measure is stocked in array Voltage, note that the library will wait for a completed conversion that takes around 200 ms@18bits

                      ^~~~~~~

MCP3424_Multichannel:80:9: error: 'class MCP342x' has no member named 'setConfiguration'

     MCP.setConfiguration(i,RESOLUTION_18_BITS,CONTINUOUS_MODE,PGA_X8);// MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

         ^~~~~~~~~~~~~~~~

MCP3424_Multichannel:80:28: error: 'RESOLUTION_18_BITS' was not declared in this scope

     MCP.setConfiguration(i,RESOLUTION_18_BITS,CONTINUOUS_MODE,PGA_X8);// MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                            ^~~~~~~~~~~~~~~~~~

MCP3424_Multichannel:80:47: error: 'CONTINUOUS_MODE' was not declared in this scope

     MCP.setConfiguration(i,RESOLUTION_18_BITS,CONTINUOUS_MODE,PGA_X8);// MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                                               ^~~~~~~~~~~~~~~

MCP3424_Multichannel:80:63: error: 'PGA_X8' was not declared in this scope

     MCP.setConfiguration(i,RESOLUTION_18_BITS,CONTINUOUS_MODE,PGA_X8);// MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                                                               ^~~~~~

C:\Users\AMP User\Documents\Arduino\libraries\MCP3424-master\Examples\MCP3424_Multichannel\MCP3424_Multichannel.ino:80:63: note: suggested alternative: 'PRIX8'

     MCP.setConfiguration(i,RESOLUTION_18_BITS,CONTINUOUS_MODE,PGA_X8);// MCP3424 is configured to channel i with 18 bits resolution, continous mode and gain defined to 8

                                                               ^~~~~~

                                                               PRIX8

MCP3424_Multichannel:82:22: error: 'class MCP342x' has no member named 'measure'

     Voltage[i-1]=MCP.measure(); // Measure is stocked in array Voltage, note that the library will wait for a completed conversion that takes around 200 ms@18bits

                      ^~~~~~~

Multiple libraries were found for "MCP342x.h"

 Used: C:\Users\AMP User\Documents\Arduino\libraries\MCP342x

 Not used: C:\Users\AMP User\Documents\Arduino\libraries\MCP3424-master

exit status 1

'class MCP342x' has no member named 'begin'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

battosai30, 2021-08-25:
What is your code ?

hairem, 2021-08-25:
I'm using the examples