battosai30/MCP3424 #10: Trying to use online examples.

niyuki, 2022-03-24:
I have been trying to build a voltage meter via Arduino and MCP3424 and I looked at this website https://wiki.dfrobot.com/MCP3424_18-Bit_ADC-4_Channel_with_Programmable_Gain_Amplifier__SKU_DFR0316_
which is including following code that is similar to your example One Shot, with your library:
<code> /* MCP 3424 version 1.2 example sketch OneShotConversion
 Written by B@tto
 Contact : batto@hotmail.fr

 In this example, one conversion per second is performed on channel 1 and 16 bits resolution.
 A new conversion has to be initiated by the user
 */

 #include <Wire.h>
 #include <MCP3424.h>

 MCP3424 MCP(0x68);   // Declaration of MCP3424 with Address of I2C

 long Voltage;

 void setup(){

   Serial.begin(9600);
   MCP.Configuration(1,16,0,1); // Channel 1, 16 bits resolution, one-shot mode, amplifier gain = 1

 }

 void loop(){

   MCP.NewConversion();    // New conversion is initiated

   Voltage=MCP.Measure();  // Measure, note that the library waits for a complete conversion

   Serial.print("Voltage = ");
   Serial.print(Voltage);
   Serial.println("uV");  // unit: microVolt

   delay (1000);

 }</code>
 
 and I guess it's a older version so I kinda tried to update it to this:
 
 <code>  /* MCP 3424 version 1.2 example sketch OneShotConversion
 Written by B@tto
 Contact : batto@hotmail.fr

 In this example, one conversion per second is performed on channel 1 and 16 bits resolution.
 A new conversion has to be initiated by the user
 */

 #include <Wire.h>
 #include <MCP342x.h>

 MCP342x MCP(0x68);   // Declaration of MCP3424 with Address of I2C

 long Voltage;

 void setup(){

   Serial.begin(9600);
     MCP.begin(0);

   MCP.setConfiguration(1,16,0,1); // Channel 1, 16 bits resolution, one-shot mode, amplifier gain = 1

 }

 void loop(){

   MCP.newConversion();    // New conversion is initiated

   Voltage=MCP.measure();  // Measure, note that the library waits for a complete conversion

   Serial.print("Voltage = ");
   Serial.print(Voltage);
   Serial.println("uV");  // unit: microVolt

   delay (1000);

 }</code>
 
 now it is uploading the script but it is not printing into the serial monitor something.. just empty
 
[image]

niyuki, 2022-03-24:
The errors first started with
MCP3424.h: No such file or directory
 so I changed it to MCP342x.h
 and then the later updates like Configuration to setConfiguration
 or MCP.Measure() to MCP.measure()
 that is how I came on the last version above

battosai30, 2022-03-24:
Hi, simply take the last version on this github ;) 1.2 is a very old one

niyuki, 2022-03-24:
I mean I downloaded the version which is here but still don't know further

battosai30, 2022-03-25:
https://github.com/battosai30/MCP3424/releases

niyuki, 2022-03-25:
Okay I downloaded the v1.9 zip file and put the folder inside the libraries folder in arduino.
When I try to upload script it drops some warnings and is "Done Uploading" but still nothing is getting printed, even if there is nothing it should print 0 or not?
The "warnings":

In file included from \Arduino\libraries\MCP3424-master\MCP342x.h:35:0,
                 from \Arduino\libraries\MCP3424-master\MCP342x.cpp:25:
\ArduinoData\packages\arduino\hardware\avr\1.8.5\libraries\Wire\src/Wire.h: In member function 'uint8_t MCP342x::getConfiguration()':
\ArduinoData\packages\arduino\hardware\avr\1.8.5\libraries\Wire\src/Wire.h:68:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int)
     uint8_t requestFrom(int, int);
             ^~~~~~~~~~~
\ArduinoData\packages\arduino\hardware\avr\1.8.5\libraries\Wire\src/Wire.h:65:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint8_t, uint8_t)
     uint8_t requestFrom(uint8_t, uint8_t);
             ^~~~~~~~~~~

Am I doing something wrong?

battosai30, 2022-03-25:
Yess it should.

What's your hardware ?
Try to add some Serial.print() like this to check if it's stuck somewhere : 

```
Serial.begin(9600); 
Serial.println("Hello");
delay(100);
MCP.begin(0);
Serial.println("begin() OK");
delay(100);
MCP.setConfiguration(1,16,0,1); // Channel 1, 16 bits resolution, one-shot mode, amplifier gain = 1
Serial.println("Config OK");
delay(100);
```

niyuki, 2022-03-25:
Ight it went great till the config part:
[image]

battosai30, 2022-03-25:
Ok this kind of problem is usually linked to i2c transmission issue as when setconfiguration() is called, it will start an i2c transaction, and if something goes wrong (main reason is bad wiring) it's halted.

Can you post a picture of your wiring ?

niyuki, 2022-03-25:
Sure, is this clear enough. I'm working just with this parts right now. I have on the left which is not on picture a wheatstone bridge and a 7 segment display. I'm just trying currently to create communication between mcp3424 and arduino mega 2560 [image]

niyuki, 2022-03-25:
Here is one from top view
[image]

niyuki, 2022-03-25:
Wait how isn't it? What did I forget

battosai30, 2022-03-25:
Nothing is wired : 

[image]

niyuki, 2022-03-25:
Ohhh okay so I tried now to power by plugging like this or switching plugs but it still can't pass to config ok. Did I plug wrong?
[image]

battosai30, 2022-03-25:
On the picture it seems okay : you need to have MCP GND -> GND of Arduino and MCP VCC -> 5V of the Arduino.

Fo your record : just remember that semiconductors hates to be exposed to inverted voltage. it's very rare that a chip has internal protection against this. So "I wire and I will see what happens" is a very bad method in electronic xD

niyuki, 2022-03-25:
Should I try it with the second MCP now with this order
[image]

battosai30, 2022-03-25:
It seems ok. Try like this, if it does not work, switch with a new MCP.
And for now, you should not connect something on any channel, we will discuss about this point after xD (one step at a time !)

niyuki, 2022-03-25:
Ok do you have time or can I contact you somewhere else?

battosai30, 2022-03-25:
Here it's ok, it adds some informations on this project ;)

niyuki, 2022-03-25:
Tried on both MCPs and still doesnt show config ok
Are u available for a call, ig thats more healthy. I'm double checking though

battosai30, 2022-03-26:
I think I would not tell you than we are discussing here xD it's so simple that I guess an obvious point is missed, so we just need to test obvious things :)

Try an i2c scan => https://playground.arduino.cc/Main/I2cScanner/ 

We will see if something is answering (it could be simple adress error)

Do you have a voltmeter ? Just check that mcp is well supllied and if i2c lines are 5V too.

niyuki, 2022-03-26:
Well it keeps printing
Scanning...
No I2C devices found
and I'm just getting more confused about this though.
I don't have a voltmeter but at school. But I got covid so I'm unable for a few days to visit school.

niyuki, 2022-03-27:
Uhm I got some nice news... I just found out that I plugged the SCL and SDA inverted so I corrected it and i2c scanner worked it is telling 0x68:
[image]

battosai30, 2022-03-28:
It's strange : 0x68 is the expected address for MCP(0)

Try MCP(0x68) in case of bug

niyuki, 2022-03-29:
MCP(0) and MCP(0x68) both doesnt get to config ok it gets stuck somehow
[image]

niyuki, 2022-03-29:
MCP(0x68) or (0) both dont go to config ok it gets stuck somehow
[image]

battosai30, 2022-03-29:
The "wiring" story is very strange : on first photo it's clearly good, you inverted them after ?

niyuki, 2022-03-29:
I don't think so. Look
[image]

battosai30, 2022-03-29:
Yes it seems ok. And with this exact wiring, i2c scan gives you 0x68 ?

niyuki, 2022-03-29:
Uhm well, somehow currently it runs but doesnt say either found or not found. See:
[image]

battosai30, 2022-03-29:
And you tried on adress 0 and 0x68 ?

niyuki, 2022-03-29:
Yes both give same results

niyuki, 2022-04-01:
Any options, ideas or what should I do now? I'm just getting more confused

battosai30, 2022-04-01:
I made a new release, install it :  https://github.com/battosai30/MCP3424/releases/tag/v1.91

Try this code and telle me the result : 

```

#include <MCP342x.h>
#include <Wire.h>

MCP342x MCP(0); 

void setup() {

Serial.begin(9600); 
Serial.println("Hello");
delay(100);
MCP.begin(0);
Serial.println("begin() OK");
delay(100);
Serial.print("Adress = 0x");
Serial.println(MCP.getAddress(),HEX);
MCP.setConfiguration(1,16,0,1); 
Serial.println("Config OK");
delay(100);

}

void loop() {}
```

niyuki, 2022-04-02:
It's showing this: 
[image]

battosai30, 2022-04-04:
Ok something is wrong : add delay(100) before setConfiguration()

niyuki, 2022-04-04:
yep going to 0x68
[image]

battosai30, 2022-04-05:
Ok ... Let's try this code : 

```
#include <Wire.h>

void setup() {

Serial.begin(9600);

Serial.println("Hello");
delay(100);
Wire.begin()
Serial.println("begin() OK");
delay(100);

Wire.beginTransmission(0x68);
Wire.write(0x28);
Wire.endTransmission();

Serial.println("Config OK");
delay(100);

}

void loop(){}
```

niyuki, 2022-04-05:
Looks good I guess
[image]

battosai30, 2022-04-05:
Whhhhhaaaatttt !! It's driving me nuts xD

Try this : 

```
#include <Wire.h>

void setup() {

Serial.begin(9600);

Serial.println("Hello");
delay(100);
Wire.begin()
Serial.println("begin() OK");
delay(100);

MCP.setConfiguration(1,16,0,1); 

Serial.println("Config OK");
delay(100);

}

void loop(){}

setConfiguration(byte channel,uint8_t resolution,uint8_t mode,uint8_t pga)
{

uint8_t cfgbyte=0;
 
//int _resolution=resolution;
//_PGA=pga;
//uint32_t divisor = 1;
//_LSB = 2048000000/(divisor<<(resolutionConvert[_resolution]-1));

cfgbyte |= (channel & 0x3) << 5;
cfgbyte |= (mode & 0x1) << 4;
cfgbyte |= (resolution & 0x3) << 2;
cfgbyte |= pga & 0x3;

Wire.beginTransmission(0x68);
Wire.write(cfgbyte);
Wire.endTransmission();

}
```

niyuki, 2022-04-05:
Did you forget to set MCP at the beginning bc it is saying 'MCP' was not declared in this scope

niyuki, 2022-04-05:
Yea already solved that but still can't run it clearly. Also tried to place the function above but this won't run so I placed it back. Currently it is:
[image]

battosai30, 2022-04-06:
I forgot a "void" before setConfiguration(byte channel,uint8_t resolution,uint8_t mode,uint8_t pga)

niyuki, 2022-04-06:
Okay it worked 
[image]

battosai30, 2022-04-06:
Oh god, I have an idea : test the multichannel example https://github.com/battosai30/MCP3424/blob/master/Examples/MCP3424_Multichannel/MCP3424_Multichannel.ino

niyuki, 2022-04-06:
Do I have to wire any cables to the other channels(currently only CH1 + and - to nothing)? This is resulting printing different values each time for same res(bits)
[image]

battosai30, 2022-04-06:
Ok the the error was a crash due to using old example syntax. So it's working :) great step !

I guess it's saturated (read the readme of the git).

Try something like : 

```
#include <MCP342x.h>
#include <Wire.h>

MCP342x MCP(0);

void setup() {

  Serial.begin(9600);  // start serial for output
  MCP.begin(0);
  MCP.setConfiguration(CH1,RESOLUTION_12_BITS,CONTINUOUS_MODE,PGA_X1); 

}

void loop(){

    Serial.print("Channel 1 : "); // print results
    Serial.print(MCP.measure());
    Serial.println(" microVolt");

    delay(500);

}
```

niyuki, 2022-04-06:
But I want to get high resolution as possible shouldnt I try 18 bits then

niyuki, 2022-04-06:
Results when I do 12bit: (at some time I just keep the wires from the battery away)
[image]

battosai30, 2022-04-06:
-1481 000 000 µV => -1.48V so, if you inverted polarity it's okay.

If you want 18 bits resolution, just change the setConfiguration() arguments

niyuki, 2022-04-06:
Okay looks good when I do 18 bits but when I remove the wire form the battery it starts throwing random values it that normal? 
[image]

battosai30, 2022-04-07:
What did you expected ?

niyuki, 2022-04-07:
I mean when I try it with a new 1.5V battery it spits value over 1.68V

battosai30, 2022-04-07:
But we agree it's not what your screenshot shows ?

niyuki, 2022-04-07:
No, I used there an old battery. This is what I meant with over 1.58V but it should be 1.5V since it was never used or not?
[image]

closed 2022-04-11 by battosai30