battosai30/MCP3424 #12: correct description in readme

scboulogne, 2022-12-01:
Hello, the information you provide in your readme on the utilities is imprecise. Here are two points that I noticed so far. 

1. long measure() retrns the result in nanovolts (nV) not in microvolts (uV). In fact, I have a multimeter connected to the adc entry and can confirm this by direct voltage measurement.
2. getRawDatas fills the 4-byte array (= one long int) with the result for just ONE channel, by simply copying it from the internal private buffer of the MCP342x object (see code). This is the last measurement obtained from the adc. When you say "for each channel" this can be wrongly interpreted.

battosai30, 2022-12-08:
Hi,

1. Good spot ;)
2. Good spot too ;)
3. 
Thanks

battosai30, 2022-12-08:
https://github.com/battosai30/MCP3424/commit/ff0937c8bdc8d8b03e69b55dd27d366b90a09292

closed 2022-12-08 by battosai30

ChrisHayward-dev, 2023-05-12:
Suggest updating the examples to report nV rather than microvolts as well...